hdl/conda-eda #6: Additional requirement for building a forked repo with Travis

RobertBaruch, 2020-02-01:
Apparently the build scripts try to access a repo under conda.anaconda.org with your github username, looking for https://conda.anaconda.org/<username>/noarch/repodata.json, which doesn't exist and fails the build. I was able to get past this by signing up to conda.anaconda.org, but maybe if you've already done that under a name other than your github username, you're hosed?

mithro, 2020-02-03:
We could use an environment variable and fall back to the github username if it isn't set?
https://docs.travis-ci.com/user/environment-variables/#defining-variables-in-repository-settings

ajelinski, 2021-03-08:
The local building is now precisely described in https://github.com/hdl/conda-eda#common-travis-ci-build-failures and it doesn't require using any CI scripts.

closed 2021-03-08 by ajelinski