hdl/conda-eda #7: Travis failed to build yosys

RobertBaruch, 2020-02-01:
```
Applying patch: '/private/tmp/conda/syn/yosys/makefile-conda-config.patch'
Trying to apply patch as-is
INFO:conda_build.source:Trying to apply patch as-is
patching file Makefile
Hunk #1 succeeded at 229 (offset 8 lines).
Hunk #2 succeeded at 519 (offset 8 lines).
Hunk #3 succeeded at 901 (offset 8 lines).
patching file tests/arch/ecp5/mux.ys
patching file tests/arch/efinix/mux.ys
Hunk #1 FAILED at 30.
1 out of 1 hunk FAILED -- saving rejects to file tests/arch/efinix/mux.ys.rej
Applying patch: '/private/tmp/conda/syn/yosys/disable-failing-ecp5-mux.patch'
Applying patch: '/private/tmp/conda/syn/yosys/skip-mux16-efinix-test.patch'
Warning: failed to download source.  If building, will try again after downloading recipe dependencies.
Error was: 
Command '['/usr/bin/patch', '-p1', '--ignore-whitespace', '-i', '/private/tmp/conda/syn/yosys/skip-mux16-efinix-test.patch']' returned non-zero exit status 1.
```

RobertBaruch, 2020-02-01:
See the build output here: https://travis-ci.org/RobertBaruch/litex-conda-eda/jobs/644817958#L1023

RobertBaruch, 2020-02-01:
Interestingly, in a fresh docker instance, it seems to patch just fine?

mithro, 2020-02-02:
Looks like as you discovered this patch was only used on OS X.

closed 2020-02-02 by RobertBaruch