hdl/conda-eda #22: Discussion: reworking travis stages

PiotrZierhoffer, 2020-06-15:
The current stage layout is not optimal, as the only purpose it serves is internal organisation.

Since the packages are now split across many repos, it doesn't really matter.

Cons of this setup:
- stages are being built sequentially even when there is no relation between packages
- even if there IS a relation between packages, it doesn't matter, as the second stage can't use the previous stage result unless running on master - otherwise the package is not uploaded to anaconda.

Two feasible solutions I see here:

- have a `Stage 1, Stage 2, Stage 3` setup
  - the division might be unclear
  - it makes master green on the first Travis pass
  - doesn't help with PRs, as the 

- ignore stages whatsoever
  - easier
  - requires a rerun of Travis on master until all dependencies are built

I am in favor of the first option. The second one can lead to strange results, like one Travis run producing `yosys` in the latest version and `yosys-plugins` built against previous `yosys`.

What are your thoughts on that?

PiotrZierhoffer, 2020-06-15:
cc @JakubJatczak @ajelinski @kowalewskijan

mithro, 2020-06-16:
I [created a Google doc](https://docs.google.com/document/d/1cj4pnacf6Vh3DySiAM3h3v6XOHUIA9WXVy7esuUyRAg/edit) to dump some data.

mithro, 2020-06-16:
Generally, yes - the stages where added more to make it easier to see the output rather than anything actually to do with stages making sense (with the one exception being the binutils / gcc-nostdc / gcc-newlib stages). You are also right that it breaks down when not running on master as the packages are not uploaded.

PiotrZierhoffer, 2020-06-17:
@mithro I added some comments  to the doc

PiotrZierhoffer, 2020-11-18:
This is long implemented

closed 2020-11-18 by PiotrZierhoffer