hdl/conda-eda #27: Package magic

mithro, 2020-06-25:
For the https://github.com/google/skywater-pdk I need https://github.com/RTimothyEdwards/magic and hence it would be good to get it packaged in conda.

PiotrZierhoffer, 2020-06-26:
https://github.com/litex-hub/litex-conda-eda/pull/28 - this is still WIP, have some troubles getting the proper configuration. Apparently magic doesn't want to compile without X (even with the relevant switch), and it seems it requires a very specific setup on host. Will continue this in few hours.

PiotrZierhoffer, 2020-06-26:
Make commands do not seem to return proper error codes for some reason

PiotrZierhoffer, 2020-06-29:
The current package builds, but tclmagic.so is not built as expected. I can't reproduce this locally, but still looking for the reason here

PiotrZierhoffer, 2020-06-29:
Found a problem with `ld` lookup. So the package builds fine. The next step - getting cairo to work, as Tim Edwards suggests you need it

PiotrZierhoffer, 2020-06-29:
Since it's merged, I close the issue. Let's open new ones if anything's wrong

closed 2020-06-29 by PiotrZierhoffer