hdl/conda-eda #69: New prjxray-db is not getting main label?

litghost, 2021-01-21:
Looks like the new prjxray-db package (see: https://anaconda.org/LiteX-Hub/prjxray-db/files) is still not getting the main label?  In particular, any package built after git hash 057e179 is good.  Any ideas why?  I original theorized that it was because of https://github.com/litex-hub/litex-conda-compilers/issues/13, but I suspect there might be another issue at play?

ajelinski, 2021-01-22:
The packages built in the workflow from `master` only get the `main` label if the whole workflow succeeds. There are a few jobs that are failing currently and I've just checked the problem doesn't come from any of the latest changes in this repository. I will see what can be done about those failures.

If you need the package immediately for testing purposes then the channel `litex-hub/label/ci-master-495615112` can be used. It will supply the very same package that would've been moved to the `main` label if the workflow had succeeded.

mithro, 2021-01-25:
@PiotrZierhoffer / @kgugala

ajelinski, 2021-01-25:
@litghost @mithro Since not having `prjxray-db` in the latest version blocks `symbiflow-arch-defs` development and currently Conda has some serious issues\*, the latest `prjxray-db` built by the CI from `master` has been moved to the `main` label manually.

\* All of the CI jobs from the latest workflow (excluding the skipped ones; https://github.com/litex-hub/litex-conda-eda/actions/runs/509194835) failed because of:
```
500 Server Error: INTERNAL SERVER ERROR for url: https://conda.anaconda.org/litex-hub/label/ci-master-509194835/linux-64/repodata.json
```

mithro, 2021-01-25:
@ajelinski Could you write up a short document about how someone does the manual move incase things like this happen in the future?

dnltz, 2021-01-25:
@ajelinski symbiflow-arch-defs needs the latest prjxray-db to work again. If I add the ci-master label I get
```
ResolvePackageNotFound: 
  - litex-hub::prjxray-db==v0.0_243_g057e179=20210125_112824
```
but the latest main label works.

ajelinski, 2021-01-26:
@mithro It's nothing special to be honest. Such move can be done with:
```
anaconda -t $ANACONDA_TOKEN move --from-label $CURRENT_LABEL --to-label main $SPEC
```
where `SPEC` needs to be `user/package/version[/filename]`.

In case of this particular package it was:
```
litex-hub/prjxray-db/v0.0_243_g057e179/linux-64/prjxray-db-v0.0_243_g057e179-20210119_091808.tar.bz2
```

ajelinski, 2021-01-26:
@dnitz This is expected since the package has been moved to the `main` label. The Anaconda doesn't seem to allow having a package with two labels simultaneously. The `litex-hub/label/ci-master-495615112` channel won't work anymore.

Now it can be installed from the plain `litex-hub` channel indeed.

dnltz, 2021-01-28:
@ajelinski - Worked for me, thanks!

ajelinski, 2021-01-29:
I believe this issue can be closed @litghost?

litghost, 2021-01-29:
> I believe this issue can be closed @litghost?

Indeed, thanks!

closed 2021-01-29 by litghost