hdl/conda-eda #142: Updating published sv2v

zachjs, 2021-10-03:
Firstly, thank you for all your work maintaining releases for HDL tools! I’m sure many appreciate the ease and simplicity this setup provides. I also periodically check in on the results at https://github.com/chipsalliance/sv-tests, which uses the package versions published here.

I've noticed that the version of sv2v under test has recently become stale (~7 weeks old). Although https://anaconda.org/LiteX-Hub/zachjs-sv2v has been updated regularly, it seems that because the other packages over at https://github.com/hdl/conda-misc have had failures, the newer versions of sv2v have not been given the `main` tag.

* Would it make sense to update the `main` tag for independent (sets of) packages, even if there are adjacent/unrelated failures?
* Regardless of the above, would it be appropriate to move sv2v from conda-misc to conda-eda? All of the other tools tested by sv-tests are packaged here in conda-eda, while sv2v is packaged in the less-active conda-misc. It seems like sv2v might be more accurately placed here, and would naturally benefit from the greater failure visibility this repo receives.

PiotrZierhoffer, 2021-10-04:
Hi @zachjs .

This is correct, we're working on bringing back conda-misc back to fitness.

Updating independent sets of packages is a great idea, but it's a maintenance pain. Frankly speaking, this should be somehow doable by GHActions, but it's not.

Either way, moving this to conda-eda makes sense. I will try it immediately.

PiotrZierhoffer, 2021-10-04:
I created a PR #143 

@zachjs I'm thinking which directory would be the best for sv2v. `conda-eda/misc`?

zachjs, 2021-10-04:
Thank you very much for working on this! It seems like `conda-eda/sv-front/` might be appropriate? That said, either folder would work, of course.

PiotrZierhoffer, 2021-10-05:
Right, `sv-front` makes sense. There was a hiccup in the CI, but hopefully it will pass now

PiotrZierhoffer, 2021-10-05:
This is merged to master, and the package is updated. I'm closing the issue, but feel free to reopen/create a new one if needed

closed 2021-10-05 by PiotrZierhoffer

zachjs, 2021-10-05:
Thank you very much for your quick work!