hdl/conda-eda #147: ETOOMANY yosys package

proppy, 2021-12-08:
Current there are the following packages distributed in various conda channel:

- for yosys:
https://anaconda.org/symbiflow/antmicro-yosys
https://anaconda.org/symbiflow/symbiflow-yosys
https://anaconda.org/symbiflow/yosys
https://anaconda.org/symbiflow/yosys-uhdm
https://anaconda.org/litex-hub/yosys
https://anaconda.org/litex-hub/symbiflow-yosys
https://anaconda.org/litex-hub/quicklogic-yosys
https://anaconda.org/litex-hub/yosys-uhdm

- for plugins:
https://anaconda.org/litex-hub/yosys-symbiflow-plugins
https://anaconda.org/litex-hub/symbiflow-yosys-plugins
https://anaconda.org/litex-hub/quicklogic-yosys-plugins
https://anaconda.org/symbiflow/symbiflow-yosys-plugins
https://anaconda.org/symbiflow/yosys-plugins
https://anaconda.org/symbiflow/uhdm-integration-yosys

https://github.com/SymbiFlow/yosys/issues/96#issue-1074001804 suggest that some of those might be obsolete, so we should make sure that this is reflected for the corresponding conda packages (see https://docs.anaconda.com/anacondaorg/user-guide/tasks/work-with-packages/#deleting-a-package).

mithro, 2021-12-08:
@PiotrZierhoffer who should own this? Someone from the SystemVerilog team maybe?

PiotrZierhoffer, 2021-12-14:
We started by removing symbiflow-yosys recipe. Arachne PNR still depends on it, but it's not trivial to replace the dependency with modern Yosys.

We'll take it step by step, phasing out different yosys builds and then removing old packages from anaconda. We're doing it as a multi-team effort, as we don't want to break anything.

proppy, 2021-12-17:
I forgot to mention https://pypi.org/project/yowasp-yosys/ which use a different approach for portability (compiling yosys as a WASM bundle).

tgorochowik, 2022-01-04:
The packages that are no longer used have been removed. So for example all the yosys packages from the Symbiflow channel.

The remaining packages are used in various places and need to stay until they are no longer needed. Let's file a few issues to see if we can start using one package everywhere.