hdl/conda-eda #149: pnr/arachne: Remove `symbiflow-yosys` dependency

ajelinski, 2021-12-14:
Building fails after making such a change (like in https://github.com/hdl/conda-eda/pull/148) but also using `yosys` instead isn't helpful. `make simpletest` fails in both cases. It's weird because according to the `test` and `simpletest` comments `yosys` shouldn't be required for the latter: https://github.com/YosysHQ/arachne-pnr/blob/master/Makefile#L101

Arachne shouldn't depend on Yosys and `symbiflow-yosys` isn't built anymore so this should be fixed.

ajelinski, 2021-12-17:
After discussions with @tmichalak there's really a question if building Arachne is still necessary.
Its source repository hasn't been updated since 2019 and it's marked as obsolete.
The latest downloads of this package were 5 months ago and each package was downloaded no more than 4 times across these last two years.

Maybe we could simply stop building Arachne and remove its recipe @mithro?

mithro, 2021-12-28:
Yeah, I think dropping Arachne is reasonable.

PiotrZierhoffer, 2021-12-30:
I removed it on master

closed 2021-12-30 by PiotrZierhoffer