hdl/conda-eda #154: Documentation build in fomu-workshop fails due to yosys not being found

umarcor, 2022-01-09:
Since a few days ago, building the Sphinx documentation of fomu-workshop is failing:

https://github.com/im-tomu/fomu-workshop/runs/4751518333?check_suite_focus=true
```
source env/bin/activate; conda config --system --add envs_dirs /home/runner/work/fomu-workshop/fomu-workshop/docs/env/envs
source env/bin/activate; conda config --system --add pkgs_dirs /home/runner/work/fomu-workshop/fomu-workshop/docs/_download/pkgs
source env/bin/activate; conda env update --name base --file ./environment.yml
Collecting package metadata (repodata.json): ...working... done
Solving environment: ...working... failed

 ResolvePackageNotFound: 
  - yosys
```

This is the environment file: https://github.com/im-tomu/fomu-workshop/blob/master/docs/environment.yml

mithro, 2022-01-10:
@PiotrZierhoffer -- Any idea what is going on here?

mithro, 2022-01-10:
Pull requests related to Yosys - https://github.com/hdl/conda-eda/pulls?q=is%3Apr+is%3Aclosed+yosys

PiotrZierhoffer, 2022-01-11:
This is an accidental victim of https://github.com/hdl/conda-eda/issues/147. 

We'll see to restore it to fitness

PiotrZierhoffer, 2022-01-12:
It seems that simply switching to use litex-hub anaconda repo should help https://github.com/im-tomu/fomu-workshop/pull/605

umarcor, 2022-01-12:
It did! Thanks!

closed 2022-01-12 by umarcor