hdl/conda-eda #160: add open_pdks package

proppy, 2022-02-01:
upstream: http://opencircuitdesign.com/open_pdks/
minimal recipe per http://opencircuitdesign.com/open_pdks/install.html:
```
./configure --enable-sky130-pdk=/path/to/skywater-pdk
make
make install
```
depends: `skywater-pdk` #159

closed 2022-02-15 by ajelinski

proppy, 2022-02-15:
@ajelinski anything else we need to do to get it uploaded?

ajelinski, 2022-02-15:
It'll be uploaded automatically by the `master-package` job once all the other jobs succeed: https://github.com/hdl/conda-eda/actions/runs/1845871121

proppy, 2022-02-15:
This does seems related to the `UNAMED` error we say in #161, see:
https://github.com/RTimothyEdwards/magic/blob/bfdf1227a704dbbfb78f42f69965ccc2c81989c1/commands/CmdE.c#L1051
https://github.com/RTimothyEdwards/magic/blob/bfdf1227a704dbbfb78f42f69965ccc2c81989c1/commands/CmdE.c#L1017

ajelinski, 2022-02-15:
Also this might be helpful:
```
2022-02-15T09:30:40.4701151Z Traceback (most recent call last):
2022-02-15T09:30:40.4701428Z   File "../common/foundry_install.py", line 2245, in <module>
2022-02-15T09:30:40.4701686Z     cwd = destlibdir,
2022-02-15T09:30:40.4701970Z   File "../common/foundry_install.py", line 277, in subprocess_run
2022-02-15T09:30:40.4702263Z     raise SystemError("".join(emsg))
2022-02-15T09:30:40.4702895Z SystemError: Command magic failed with exit code: -11
2022-02-15T09:30:40.4703391Z   magic -dnull -noconsole $SRC_DIR/sky130/sky130A/libs.ref/sky130_fd_sc_hd/spice/generate_magic.tcl
```

BTW There are a lot of these `Cell "sky130_fd_sc_hd__fill_4" is used but not defined in this file.` errors even in the successful job so they might not be really relevant. I guess this " Rewinding input.  Cells may have been instanced before they were defined." must be true but then I'm not sure why this isn't just a warning.

Is it a problem observable outside this repository's CI?

proppy, 2022-02-15:
I suspect the actual fatal error is:
```
  Please name the cell before extracting.
```
as it seems to be a fatal error according to the code path I linked in https://github.com/hdl/conda-eda/issues/160#issuecomment-1040144529

ajelinski, 2022-02-15:
Oh yeah, it seems you're right.
I've restarted the CI to have the package uploaded.

I guess we'll see in the coming days how it works in the nightly CI but I'm afraid I'll have to disable the job if it turns out to be failing too often. Unfortunately a single failing job prohibits other packages from being updated.

PR #164 (2022-02-15): misc: add open_pdks
Fixes #160
Depends on #161 

- package  as `open_pdks.sky130a`
- mimic `make timing` in build script to leverage conda environment for build dependencies

@mithro and @PiotrZierhoffer for review
@olofk @RTimothyEdwards  should we expose more combination of libs as different subpackages?