hdl/conda-eda #170: conda-build fails when there is no tag available

proppy, 2022-02-09:
Most of the recipes starts with a similar boilerplate to derive the version number from the current git tag:
```
{% set version = '%s_%04i_%s'|format(GIT_DESCRIBE_TAG|replace('v', '') or '0.X.X', GIT_DESCRIBE_NUMBER|int, GIT_DESCRIBE_HASH or 'gUNKNOWN') %}
```
Which seems to fails if the corresponding git repo has no tags:
https://github.com/hdl/conda-eda/runs/5108676826?check_suite_focus=true

Using `|default()` instead or `or` seems to workaround this issue:
https://github.com/hdl/conda-eda/pull/168/commits/8d74975bdf648f059211eac02d6a432df0f4690f

But we could also be lookin at using `GIT_DESCRIBE_TAG` (documented  at https://docs.conda.io/projects/conda-build/en/latest/user-guide/environment-variables.html#git-environment-variables) to simplify the boilerplate.

mithro, 2022-02-09:
This sounds like you didn't use the conda-build-prepare stuff to build the package which makes sure the git repository has a tag?

proppy, 2022-02-09:
I assumed the CI does run `conda-build-prepare`, see:
https://github.com/hdl/conda-eda/runs/5108676826?check_suite_focus=true
```
  Building wheels for collected packages: conda-build-prepare
    Building wheel for conda-build-prepare (setup.py): started
    Building wheel for conda-build-prepare (setup.py): finished with status 'done'
    Created wheel for conda-build-prepare: filename=conda_build_prepare-0.1.1-py3-none-any.whl size=21414 sha256=ed6815ef456403d469a82262bf8bb5977dd458c14b94ad10f7007d4a794596ca
    Stored in directory: /tmp/pip-ephem-wheel-cache-x52rj20v/wheels/7d/15/5a/8d1ebdbef1db8143276ef6a3b9ea2be642a9211b2fdb8964f0
  Successfully built conda-build-prepare
  Installing collected packages: ruamel.yaml.clib, MarkupSafe, ruamel.yaml, jinja2, conda-build-prepare
  Successfully installed MarkupSafe-2.0.1 conda-build-prepare-0.1.1 jinja2-3.0.3 ruamel.yaml-0.17.20 ruamel.yaml.clib-0.2.6
  Cloning into '/home/runner/work/conda-eda/conda-eda/workdir/git-repos/xls'...
  Already on 'main'
  ERROR conda.cli.main_run:execute(34): Subprocess for 'conda run ['conda', 'render', '-f', '/home/runner/work/conda-eda/conda-eda/workdir/recipe/rendered_metadata.yaml', '/home/runner/work/conda-eda/conda-eda/workdir/recipe']' command failed.  (See above for error)
  Error: Failed to render jinja template in /home/runner/work/conda-eda/conda-eda/workdir/recipe/meta.yaml:
  'GIT_DESCRIBE_TAG' is undefined
  ```

ajelinski, 2022-02-14:
This isn't really the problem of a missing tag. The tag is definitely added by `conda-build-prepare`:
```
  Rewriting tags in "/home/runner/work/conda-eda/conda-eda/workdir/git-repos/xls"...
  
  Successfully tagged '3875ebb5b8ed1036253dc2757e667541fd70ec99' object as 'v0.0'
  Rendering package metadata, please wait...
```

`conda render` and `conda build` give this misleading error for many git-related problems, such as the ones for applying patches.

I've looked into the problem and it's quite weird. In short: ` conda build/render` apply patches with `git am -3 --committer-date-is-author-date` if the first line is `From: <SHA>` and with `patch -p` otherwise. Git has some unexpected problem to apply `add-dist-rule.patch`. This patching problem seems to later prevent Conda from calling `git describe` which then is a problem for Jinja as it doesn't have a `GIT_DESCRIBE_TAG` used in the `meta.yaml`.

It turns out it's enough to comment the first line of the patch or its whole `git am` header because `patch -p`, which is then used, works well.

In all our Conda repositories only `libxml2` contains patches with `git am` headers but there the source is an archive instead of a git repository so perhaps something's broken for git repositories + git-am patches currently in Conda.

ajelinski, 2022-03-30:
Closing this issue since the problem with `xls` was indeed caused by a patch.

Given that `conda-build` successfully renders the recipe (which includes applying patches to the source), `conda-build-prepare` properly provides version even if there are no tags in the repository (eg: [prjoxide](https://github.com/gatecat/prjoxide)).

closed 2022-03-30 by ajelinski