hdl/conda-eda #171: Add ghdl and ghdl-yosys-plugin

shenki, 2022-02-11:
In order to use VHDL with the yosys flows, we need ghdl and the ghdl yosys plugin to be part of the environment.

https://github.com/ghdl/ghdl-yosys-plugin

https://github.com/ghdl/ghdl.git

kgugala, 2022-02-11:
@umarcor can you take a look at this?

umarcor, 2022-05-12:
@proppy @PiotrZierhoffer would you know how to [customize a compiler](https://docs.conda.io/projects/conda-build/en/latest/resources/compiler-tools.html#customizing-the-compilers) to enable Ada support in GCC?

/cc @mithro

proppy, 2022-05-12:
@umarcor wouldn't https://github.com/conda-forge/staged-recipes/issues/14692 also help?

umarcor, 2022-05-12:
@proppy, I believe that the main purpose is to support it on Linux, because all the tools are expected to be available, while some might be missing on Windows or macOS.

MSYS2 packages on Windows are known to work (one of my main workstation is a Windows 10 with MSYS2).

proppy, 2022-05-12:
@umarcor ah yes, what I meant was that having the ada compiler available in conda-forge could enable us to build a recipe for this.

mithro, 2022-05-12:
https://docs.conda.io/projects/conda-build/en/latest/resources/compiler-tools.html#using-your-customized-compiler-package-with-conda-build-3

nobodywasishere, 2023-01-07:
Also found a gcc ada compiler that runs on M1 macs
https://github.com/simonjwright/distributing-gcc/releases/tag/gcc-12.2.0-aarch64

proppy, 2023-03-22:
It seems that there is https://github.com/hdl/conda-eda/search?q=ghdlsynth in the repo, how is that related to this effort?

proppy, 2023-03-22:
/cc @xobs

umarcor, 2023-03-23:
@proppy see https://github.com/hdl/conda-eda/commits/master/syn/ghdlsynth. I guess it never worked because there was never an Ada compiler available. So, the recipe/metadata are good enough, but a build time dependency is missing in the environment.