hdl/conda-eda #184: use dedicated GCP runners

proppy, 2022-02-24:
We should consider using https://antmicro.com/blog/2021/08/open-source-github-actions-runners-with-gcp-and-terraform/ to get quicker feedback on the CI jobs.

proppy, 2022-02-24:
yes! https://console.cloud.google.com/?project=github-hdl

mithro, 2022-02-27:
@umarcor / @kgugala - Should have pending invites to manage the `github-hdl` organization.

proppy, 2022-12-13:
@PiotrZierhoffer @ajelinski @kgugala how can I help to setup dedicated runner for conda-eda? per https://github.com/hdl/conda-eda/issues/210#issuecomment-1345178060 this will become necessary for bigger packages like Xyce.

kgugala, 2022-12-13:
@proppy We can setup the infrastructure. Do you know what machines will be needed?

proppy, 2022-12-13:
There are currently 64 package building jobs being run concurrently on every commit.

The longest job (XLS) takes around 2 hours and dominate the total build time.

Assuming you can't have multiple runner running on a single node what about starting with 4x n2-standard-32?

proppy, 2022-12-13:
> you can configure this in the yaml file

per job or per workflow?

mithro, 2022-12-13:
per job I think?

umarcor, 2022-12-14:
@proppy what's the issue with building Xyce on default runners? We build it in hdl/containers and it takes less than 2h, which is far below the limit. Are you cross-compiling it for architectures other than x64?

proppy, 2022-12-21:
@umarcor this was based on a conversation with @cbalint13 here: https://github.com/hdl/conda-eda/issues/210#issuecomment-1345178060

cbalint13, 2022-12-21:
> @umarcor this was based on a conversation with @cbalint13 here: [#210 (comment)](https://github.com/hdl/conda-eda/issues/210#issuecomment-1345178060)

@umarcor , @proppy 

Trilinos, takes full ~15h (Release, no-debug), all 3x complete builds {native, openmpi, mpich}:
   * receipt: https://copr-dist-git.fedorainfracloud.org/cgit/rezso/HDL/trilinos.git/tree/trilinos.spec
   * builds: https://copr.fedorainfracloud.org/coprs/rezso/HDL/build/5122252

Xyce, takes full ~1h (Release, no-debug), all 3x complete builds {native, openmpi, mpich}:
   * receipt:  https://copr-dist-git.fedorainfracloud.org/cgit/rezso/HDL/xyce.git/tree/xyce.spec
   * builds:  https://copr.fedorainfracloud.org/coprs/rezso/HDL/build/5160728/

Automated builds having full possible flags/features, except CUDA for now (coming soon).
All builders {aarch64, x86-64, ppc64le} are native, no cross compilation, ram + rootramfs is ~150Gb.

proppy, 2023-01-17:
@AdamOlech configured the custom runner and posted a PoC here:
- https://github.com/hdl/conda-eda/blob/custom-runners-test/.github/workflows/Build.yml
- https://github.com/hdl/conda-eda/actions/runs/3911464894/jobs/6684888111

@ajelinski @PiotrZierhoffer should we migrate all the job at once? or incrementally?

PiotrZierhoffer, 2023-01-17:
We are working on moving the whole workflow. It should be easier than dividing the current one, as dependencies should be generally the same for everything.

proppy, 2023-02-24:
Thanks for doing this!

@PiotrZierhoffer @ajelinski @AdamOlech

closed 2023-02-24 by proppy