hdl/conda-eda #191: nextpnr-xilinx: `bbaexport.py` failure

ajelinski, 2022-03-07:
The CI fails to build `nexpnr-xilinx`: https://github.com/hdl/conda-eda/runs/5446064939?check_suite_focus=true#step:4:1613
```
  + pypy3 xilinx/python/bbaexport.py --device xc7a35tcsg324-1 --bba /home/runner/work/conda-eda/conda-eda/workdir/conda-env/conda-bld/nextpnr-xilinx_1646646241146/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_pla/share/nextpnr-xilinx/xc7a35tcsg324-1.bba
  Traceback (most recent call last):
    File "xilinx/python/bbaexport.py", line 350, in <module>
      main()
    File "xilinx/python/bbaexport.py", line 31, in main
      d = import_device(args.device, xraydb_root, metadata_root)
    File "/home/runner/work/conda-eda/conda-eda/workdir/conda-env/conda-bld/nextpnr-xilinx_1646646241146/work/xilinx/python/xilinx_device.py", line 458, in import_device
      with open(prjxray_root + "/" + fabricname + "/tilegrid.json") as gf:
  FileNotFoundError: [Errno 2] No such file or directory: '/home/runner/work/conda-eda/conda-eda/workdir/conda-env/conda-bld/nextpnr-xilinx_1646646241146/work/xilinx/python/../external/prjxray-db/artix7/xc7a35t/tilegrid.json'
```

FWIW the breaking change was introduced to adjust to prjxray changes: https://github.com/gatecat/nextpnr-xilinx/commit/565588a69ea95a52f7c7592f4ed81d9bef6cfb60 so I guess it should also be looked into.

I'm going to disable `nextpnr-xilinx` job to fix the CI for other packages.

proppy, 2022-03-07:
manually tagged latest build of the following package with `main` as a workaround:
- magic
- netgen
- yosys
- openroad
- open_pdks.sky130a

proppy, 2022-03-07:
only set the latest build of `yosys` with main, as I wasn't sure if we wanted to publish all the intermediate version (/cc @PiotrZierhoffer ?)

PiotrZierhoffer, 2022-03-09:
@proppy not really, we're mostly interested in the latest build