hdl/conda-eda #198: Conda-forge and Viper

curtisma, 2022-03-14:
Have you considered submitting these recipes to Conda-forge rather than having your own channel?   It could handle all the build infrastructure and the recipes would be maintained by the tool owners and/or anyone else interested in helping maintain them.  They also have a well documented process for creating new recipes which could make it easier to support new tools.

I've proposed this approach as part of the [Viper circuit design environment ](http://www.cascode-labs.org/viper/)and [viper-forge](http://www.cascode-labs.org/viper-forge/).

mithro, 2022-03-14:
Sounds like you just invented the same thing I did --> [bit.ly/edda-conda-eda-spec](https://bit.ly/edda-conda-eda-spec)

mithro, 2022-03-14:
See also  https://j.mp/edda-status

mithro, 2022-03-14:
Also some stuff at https://docs.google.com/document/d/11XFnJ0ExBgE1pMQksw0rQerAZo3F83AVIu2YK1pbg1k/edit

curtisma, 2022-03-14:
haha, cool, want to work together?

curtisma, 2022-03-14:
I work on more analog / RF IC design but there should be a lot of overlap.

mithro, 2022-03-14:
BTW Have you seen Google's SkyWater open source PDK (https://github.com/google/skywater-pdk) and the associated no-cost shuttle program for open source designs @ efabless.com?

mithro, 2022-03-14:
It seems like our work is pretty complimentary too! Our focus has been heavily on the CPU / digital (starting with FPGAs + cross compilers) while your focus has been more on the analog side.

curtisma, 2022-03-14:
Yeah, I've heard about the Sky water shuttles.  I've been considering putting together and submitting a design.

What's the best way to collaborate?  Do you want to have an online meeting or discuss on one of our GitHub discussion boards perhaps?

proppy, 2022-03-15:
@curtisma I just happen to have started a PR to add a subset of those package to conda-forge as part of #193, see https://github.com/conda-forge/staged-recipes/pull/18295

curtisma, 2022-03-15:
@proppy Ok, awesome, do you need help with it?

proppy, 2022-03-17:
@curtisma rebased the PR and it nows pass the CI for linux: https://github.com/conda-forge/staged-recipes/pull/18295, do you want to take a look before I ping the conda-forge folks about it?

curtisma, 2022-03-18:
Sure, I can take a look tonight.  That's what I had guessed when I saw it previously

curtisma, 2022-05-25:
Awesome, thanks @proppy !  I'll try it out this evening.

curtisma, 2022-05-25:
@proppy Now we need to package some type of schematic editor such as xcircuit.

curtisma, 2022-05-26:
Yeah, it's on my to-do list.
I've put together a list of packages we'd like available initially for an open source IC design conda environment:

https://github.com/cascode-labs/viper-forge/issues/1

proppy, 2022-05-26:
@mithro https://anaconda.org/conda-forge/ngspice :)

proppy, 2022-05-26:
for klayout also see: https://github.com/flaport/condalayout/issues/1#issuecomment-1075152682