hdl/conda-eda #205: provide example usage using colab notebooks

proppy, 2022-04-18:
It would be nice to have a collection of https://colab.research.google.com/ that demonstrate usage of the conda-eda packages to perform basic EDA talks.

We could seed it with some of the examples below:
https://colab.research.google.com/gist/proppy/26525ff7d9810247759a8d3ff3c41dcb/openlane-with-conda-eda.ipynb
https://colab.research.google.com/gist/proppy/b8a35e01e22c9685a79fa144e0df3f36/mosfet_simulation.ipynb
https://colab.research.google.com/drive/1ww2Upi_UD_I9Z8Xwt-1vFwsysNEiznMu

@mithro suggested using a separate repo `notebooks-eda`.

proppy, 2022-05-11:
@mgielda @umarcor can you weight-in if you think that's a good idea to create this repo?

I'd be cool to get a community maintained location for this, and I'm happy to seed it with the content I posted before on https://gist.github.com/proppy/public.

proppy, 2022-05-11:
@mattvenn also mentioned some recent interest from the https://www.zerotoasiccourse.com/ crowd.

mithro, 2022-05-11:
@umarcor - Can you create `notebooks-eda` and give @proppy admin privileges?

umarcor, 2022-05-11:
As commented through e-mail some months ago, I don't think it's worth creating a repo for the purpose of hosting notebooks only. There are existing locations which can host such content:

- [hdl.github.io/packages](https://hdl.github.io/packages/) is expected to contain examples and references about how to use the multiple packaging solutions with simple and not so simple FPGA and ASIC examples. It can host all the examples/flows that don't fit in f4pga-examples, such as using containers (https://github.com/hdl/packages/blob/main/.github/workflows/containers-conda-f4pga.yml, https://github.com/chipsalliance/f4pga-examples/issues/227), edalize, notebooks... Therefore, notebook examples do fit in there. We can integrate them into the sphinx site ([nbsphinx.readthedocs.io](https://nbsphinx.readthedocs.io)), or we can host them in a subdir using some other "engine" (and link them from sphinx).

- There are six conda related repos in this org, which don't have any doc site and some of them have an outdated README. If the notebooks are focused on using Conda (and not other packaging solutions) those would fit in one of them. For instance, the README from [hdl/conda-eda](https://github.com/hdl/conda-eda) can be updated to include the links to the colabs, or converted to an Sphinx site (and the notebooks can be added in there).

mithro, 2022-05-11:
As an end user, I would not expect to find notebooks in either `hdl/conda-eda` or `hdl/packages`, I **_might_** expect to find notebooks in `hdl/examples` or `hdl/demos` but there would not be anything which clearly indicates that such a possibility even exists. A repository like `hdl/notebooks-eda` is pretty clear what I will find in it. While we are pretty use to the idea of using open source EDA tools inside a notebook, most people find that extremely unusual and wouldn't even think to look for it.

I **_agree_** that we can use notebooks inside other repositories to help with documentation and demos, but having a nice gallery which a user can browse which shows the type of things that can be done is very useful. It can also be a way to get people to introduced to this GitHub organization. Currently we are just linking to random gists under @proppy's username. If we don't end up with `hdl/notebook-eda` the alternative is going to be something like `chipsalliance/notebook-eda` or `The-OpenROAD-Project/notebook-eda`.

umarcor, 2022-05-12:
For visibility purposes:

- We can add a `.github/profile/README.md` which points to the most relevant content (containers, conda, notebooks, examples, etc.).
- The docs of hdl.github.io/packages can be served at hdl.github.io and hdl.github.io/notebooks can point to the content we want regardless of the actual location.

As an end user, I would not expect to find examples, demos or notebooks in org hdl. The main purpose of this org has been gathering:

- Tool building and packaging recipes, smoke tests and CI (for Conda, containers, MSYS2, Bazel, Termux...).
- References about existing projects/tools.
- Data about boards, devices, etc.

Examples for end-users are available in chipsalliance/f4pga, librecores/fusesoc/edalize, fomu, NEORV32 setups, Litex, zerotoasic...

I'm good with adding some example/demo content to the repos in this org, provided that some of them are lacking any documentation. However, I think that other orgs are a better fit for an example focused repository.
Nonetheless, since the existing notebooks are using conda packages only, we can start with having a `notebooks` subdir in this repo and move it later if it finds a better location in some other org.

umarcor, 2022-05-12:
Is this the logo you want to use for the notebooks? https://simpleicons.org/icons/googlecolab.svg 

Or shall it be https://simpleicons.org/icons/jupyter.svg ?

proppy, 2022-05-23:
> However, I think that other orgs are a better fit for an example focused repository.

we discussed this topic during https://chipsalliance.org/workgroups/#:~:text=Mailing%20list-,Tools%20Workgroup,-The%20Tools%20workgroup w/ @kgugala and @pgielda and agreed that https://github.com/chipsalliance/silicon-notebooks could provide a good home for this project has various member of chipsalliance expressed interest into leveraging notebook as a way to lower the barrier of entry to the open silicon ecosystem.

closed 2022-05-23 by proppy