hdl/conda-eda #210: Add Xyce simulator as a package

mithro, 2022-05-06:


proppy, 2022-05-24:
First we need to identify:
- upstream
- build documentation
- dependencies

umarcor, 2022-05-24:
@proppy see:

- https://github.com/hdl/containers/blob/main/debian-bullseye/xyce.dockerfile
- https://github.com/hdl/containers/issues/56
- https://github.com/google/skywater-pdk/issues/186
- https://github.com/Xyce/Xyce/blob/master/INSTALL.md

umarcor, 2022-05-24:
@proppy, note that you need to use version 12.12.1 of Trilinos (https://github.com/hdl/containers/blob/main/debian-bullseye/xyce.dockerfile#L54). Fortunately, it is available: https://anaconda.org/conda-forge/trilinos/labels.

saicharan0112, 2022-09-01:
Any progress on this? Is there something I can help here?

cbalint13, 2022-11-12:
> Any progress on this? Is there something I can help here?

@saicharan0112

If you want it for fedora/rhel you can have latest (including xyce):
https://copr.fedorainfracloud.org/coprs/rezso/HDL

Also the repo can be invoked within Docker for convenience, a short demo:
https://gist.github.com/cbalint13/1de29fbe77e8dedddd3861b5f49a1835

saicharan0112, 2022-12-07:
Thanks, @cbalint13 .

saicharan0112, 2022-12-10:
This is very interesting and useful for Fedora/RHEL users, @cbalint13 . I wish we had something similar for most used distros (ubuntu or debian in general). I am seeing conda packages as more like a platform independent way for installations. Not sure if the conda packaging mechanism keeps track on dependencies like its done with Fedora and the conda-forge.

cbalint13, 2022-12-10:
@saicharan0112 

> This is very interesting and useful for Fedora/RHEL users, @cbalint13 . I wish we had something similar for most used distros (ubuntu or debian in general). 

* Native packages are doable but there is no build-farm like COPR (free) that can target ubuntu / debian world.
* The same is with conda (no farm), so the burden of recipes is in charge of the users having their compile times.
 
> I am seeing conda packages as more like a platform independent way for installations. Not sure if the conda ?packaging mechanism keeps track on dependencies like its done with Fedora and the conda-forge.

* There is a very rigorous QA assurance of such native packages (.rpm/.deb) are distro compliant on their own.
* One can easily compose his own docker images out of these rhel/fedora packages (without any compile times).

proppy, 2022-12-10:
> @saicharan0112 <https://github.com/saicharan0112>
>
> This is very interesting and useful for Fedora/RHEL users, @cbalint13
> <https://github.com/cbalint13> . I wish we had something similar for most [...]
Small clarification:

conda-forge actually has a farm on azure (we rely on it for the magic
package: https://github.com/conda-forge/magic-feedstock)

And this repo has some continuous integration running on GitHub actions
that upload packages to a dedicated channel: https://anaconda.org/LiteX-Hub

But I agree the Q&A is way less rigorous than a distro based farm like you
have with fedora.

cbalint13, 2022-12-10:
@proppy 

> Small clarification: conda-forge actually has a farm on azurehttps://github.com/conda-forge/magic-feedstock) 

That is interesting. To my understanding, the very builders are kind of (re)purposed azure, etc. instances ?

> (we rely on it for the magic package
>And this repo has some continuous integration running on GitHub actions that upload packages to a dedicated channel: https://anaconda.org/LiteX-Hub

Magic / Litex (also) are small ones, many HDL/EDA build space reaches >100 Gbyte and some >12h of build times.
What resources would be on these instances for regular builds of bigger (i.e. Xyces / Trilinos, two good examples) ?

proppy, 2022-12-13:
> That is interesting. The very builders are kind of (re)purposed azure, etc. instances ?

I think they are using https://azure.microsoft.com/en-us/products/devops/pipelines/.

> What resources would be on these instances for regular builds of bigger (i.e. Xyces / Trilinos, two good examples) ?

For conda-eda specifically, we discussed seting up custom github runner on dedicated Google Compute instance in #184.

saicharan0112, 2022-12-13:
@proppy If it is of any help while building conda package for xyce - https://github.com/Xyce/Xyce/discussions/4#discussioncomment-169255

This works very smooth. I did this on 3 machines.

proppy, 2022-12-13:
@saicharan0112 Thanks! I think we're doing something similar here already https://github.com/hdl/conda-eda/pull/263/files#diff-131443e74262f22bf247f65d935f723616400829a221a5437dac94721f4d1b40