hdl/conda-eda #215: add coriolis package

proppy, 2022-05-24:
Upstream: https://gitlab.lip6.fr/vlsi-eda/coriolis
Build documentation: https://gitlab.lip6.fr/vlsi-eda/coriolis/-/blob/devel/README.rst#L38
Dependencies:
* Python 3
* cmake
* boost
* bison & flex
* Qt 4 or 5
* libxml2
* RapidJSON

Seems like qt is a hard-dep: https://gitlab.lip6.fr/vlsi-eda/coriolis/-/blob/devel/bootstrap/cmake_modules/FindBootstrap.cmake#L211

mithro, 2022-05-24:
While I would love to see this, I would say it is a much lower priority than Xyce / OpenLane / etc.

proppy, 2023-07-24:
New upstream repos:
* https://github.com/lip6/coriolis
* https://github.com/lip6/alliance
* https://github.com/lip6/alliance-check-toolkit
* https://github.com/lip6/tas-yagle