hdl/conda-eda #222: add charon package

proppy, 2022-06-30:
Upstream:
https://charon.sandia.gov/
Dependencies:
https://charon.sandia.gov/third-party-libraries/
License:
GPLv3
Build recipe:
```
build_charon.py
```

thomasdorch, 2022-08-29:
It's still in progress, but I am working on a [conda package for Charon](https://github.com/thomasdorch/charon-conda).

proppy, 2022-08-29:
@thomasdorch do you want to co-maintain this together in this repo?

thomasdorch, 2022-08-30:
@proppy I am hoping to get it into conda-forge at some point, would having it in the conda-eda repo conflict with that at all? I am still sort of figuring out how the conda ecosystem works but if there isn't a conflict, I'd be happy to co-maintain it here.

proppy, 2022-08-31:
@thomasdorch I'd be happy to help with the conda-forge package as well!

I'm relatively new to conda as well, but here are some thoughts we put together on conda-eda vs conda-forge: https://github.com/hdl/conda-eda/issues/193#issuecomment-1084035801, feel free to contribute yours to the conversation!

thomasdorch, 2022-08-31:
Thanks for linking that discussion @proppy! I'm down to co-maintain it in conda-eda, and would definitely appreciate your help in getting it into conda-forge. 
So would including it in conda-eda mean that all of the dependencies that I'm currently pulling from conda-forge (hdf5, netcdf, etc.) need to be compiled as part of the build.sh script?

thesourcerer8, 2022-09-01:
I packaged Charon for Docker, perhaps that helps you too: https://github.com/thesourcerer8/OpenSourceTCAD/