hdl/conda-eda #231: add SODA package

proppy, 2022-08-08:
Upstream:
https://gitlab.pnnl.gov/sodalite
Dependencies:
- LLVM
- MLIR
License:
https://gitlab.pnnl.gov/sodalite/soda-opt/-/blob/main/LICENSE.txt
Build recipe:
https://gitlab.pnnl.gov/sodalite/soda-opt#how-to-build

agostini01, 2022-08-09:
Hi @proppy,

Thank you for writing this issue.

I am one of the maintainers of the soda-opt project. Please, let me know of any questions you have.

Also, please, let me know if there is a specific structure (or example) you would like us to follow. I can adapt or create specific build scripts to address any issues regarding implementing this feature.

proppy, 2022-09-07:
@agostini01 sorry for the late followup, the first thing we need to identify is if all SODA dependencies are available in https://anaconda.org/.

Found:
- https://anaconda.org/main/llvm
- https://anaconda.org/main/mlir

Are there any other dependencies needed?

agostini01, 2022-09-08:
Hello @proppy ,
No problems at all.

Just want to confirm that you mean post installation dependencies. Is it correct?

- SODA-OPT (compiler frontend) is compiled statically.
- Bambu HLS (compiler backend) depends on `verilator libbdd-dev g++-7-multilib` and any compiler that was present during compilation of Bambu (it checks if gccX and clangX are in the path and compiles plugins to enable their usage as frontends). We usually try to ship it with clang12 and gcc7.
- We may also need tf-mlir-translate, tf-opt (or other similar tools) for translating into MLIR

proppy, 2022-09-08:
> SODA-OPT (compiler frontend) is compiled statically.

For SODA-OPT, even if it's compiled statically when building it from source, we should try for this package to depends as much of possible on the existing conda packages for build and runtime dependencies.

> Bambu HLS (compiler backend) depends on verilator libbdd-dev g++-7-multilib

For Bambu HLS, let's track those requirements in https://github.com/hdl/conda-eda/issues/230

> We may also need tf-mlir-translate, tf-opt (or other similar tools) for translating into MLIR

Is that a build or a runtime hard dependencies? (SODA doesn't work in it's not there), or just a recommended packages nice to have installed around when using SODA?

agostini01, 2022-09-08:
| Is that a build or a runtime hard dependencies?
They are not hard dependencies, but the MLIR IR syntax changes from time to time. So that the MLIR generated by `tf-mlir-translate` in the past may not be compatible with MLIR IR accepted by `soda-opt`/`mlir-opt` now. In our compilation pipelines, I try to keep them updated based on the same/similar version of the main branch in the llvm-project, which is now beyond the stable versions you suggested (LLLVM14 seem to be the latest).

|For Bambu HLS, let's track those requirements in https://github.com/hdl/conda-eda/issues/230
Got it. These were runtime dependencies. I see you have compile time dependencies in https://github.com/hdl/conda-eda/issues/230

|We should try for this package to depends as much of possible on the existing conda packages for build and runtime dependencies
Understood. So, if I create a clean ubuntu image, install conda+new environment, and start issuing `conda install <package-name>` instead of `apt install <package name>` until my dependencies are met, would the final list of packages be what you are looking for?
What if I cannot find a package in conda that only exist in apt repositories?