hdl/conda-eda #235: integrate with efabless patches in open_pdks packages

proppy, 2022-08-19:
It seems that that efabless is maintaining patches on top of the default `open_pdks` repo for better integration with OpenLane: https://github.com/efabless/open_pdks/.

This was discussed as https://github.com/chipsalliance/silicon-notebooks/pull/9 as part of some delta in the default values that OpenLane expect to be defined for a given PDK variant. 

We could integrate those locally patch in the conda packages, until they are contributed upstream or target https://github.com/efabless/open_pdks/ as the upstream repository.

proppy, 2022-08-19:
@donn will that be a long-lived fork? or should I just apply the patch https://github.com/efabless/open_pdks/commit/44a43c23c81b45b8e774ae7a84899a5a778b6b0b locally to the package until it get merged upstream?

donn, 2022-08-19:
This is a long-lived fork. It's the repository of open_pdks used by volare and OpenLane. We do merge things upstream and merge things from upstream, but this overall gives me more control over variables, including OpenLane configuration variables and real-life variables such as Tim's availability.

So if your intent for `open_pdks` is to be a canonical installation, use Tim's repo with a patch applied, if your intent is to support OpenLane, use the Efabless repo.

proppy, 2022-09-27:
The target repo has been reconciled with the upstream one.