hdl/conda-eda #260: upgrade to python 3.8

proppy, 2022-12-04:
python 3.7 is EOL'ed next year, and colab has recently updated to 3.8:
https://twitter.com/GoogleColab/status/1598059818769870849

The colab fallback Python 3.7 runtime will only available until mid-December, see https://medium.com/google-colab/colab-updated-to-python-3-8-4922f9970a72

So until then we should make sure to have 3.8 versions of the packages available.

mithro, 2022-12-04:
We should make sure to have packages for Python 3.9, 3.10 and 3.11

erendn, 2023-01-24:
As I mentioned in https://github.com/KLayout/klayout/issues/1263, installing klayout gives python version conflict with other tools using python 3.8.

proppy, 2023-02-24:
Duplicate with #72.

closed 2023-02-24 by proppy