hdl/conda-eda #276: add openvaf package

proppy, 2023-01-05:
### Upstream

https://github.com/pascalkuthe/openvaf

### Dependencies

- Rust
- LLVM
- https://crates.io/crates/open-vaf/0.4.2/dependencies

### Build

- https://github.com/pascalkuthe/OpenVAF#building-openvaf

### License

GPLv3

proppy, 2023-01-05:
would it make sense to also package [VerilogAE](https://github.com/pascalkuthe/OpenVAF/tree/master/verilogae) and [melange](https://github.com/pascalkuthe/OpenVAF/tree/master/melange)?

metroid120, 2023-01-05:
I would package VerilogAE if modeling engineers are working with conda-eda, else not. This tool allows to use Verilog-A defined model equations in Python, so the target audience is only modeling engineers.

Regarding Melange, I suggest to get @pascalkuthe `s opinion.

pascalkuthe, 2023-01-05:
Thank you for starting this initiative a couple of comments:

* Melange is still an early prototype that is not yet ready for widespread use. I currently use melange for testing OpenVAF but we currently don't have the resources to push it further so I wouldn't package it.
* VerilogAE is a python package that uses (parts) of OpenVAF internally that has been released for a while and is already seeing quite a bit of use. It's published as a prebuild wheel on pypi (that has no external dependencies). I am not sure what the goals of conda-eda are and if it makes sense to repackage VerilogAE if the wheels are already available on pypi.
* OpenVAF depends exactly on LLVM 14, older and newer mayor versions will not work, it only depends on LLVM itself but also on `LLD` which is part of the LLVM monorepo (as mentioned in the build instructions)
* The binary available on [openvaf.semimod.de](https://openvaf.semimod.de) is fully statically linked so there are no runtime dependencies. A version for windows is also available. Building LLVM for OpenVAF can be tricky so it might sense to just use the official pre-build binary instead of compiling OpenVAF yourself.
* We use a docker image to build openvaf that has all relevant dependencies available: https://github.com/pascalkuthe/ferris-ci/pkgs/container/ferris_ci_build_x86_64-unknown-linux-gnu. The associated repo may help with setting up the build
* You linked `crates.io` for dependencies. I published an extremely old version of openvaf there years ago to reserve the name. OpenVAF is not published to crates.io and will probably never be published to `crates.io` as it's split across many  crates. To build OpneVAF use the source-code from the release tag instead. The dependencies shown on crates.io are completely inaccurate as a result. That shouldn't matter much however as `cargo` automatically downloads, compiles and statically links any dependencies . Building and packaging all of these separately is likely a futile effort as OpenVAF is comprised of about 138 crates (around 30 of which are part of the OpneVAF repo the rest are crates.io dependencies) plus about 40 extra crates required for VerilogAE, melange and test infrastructure.

proppy, 2023-01-06:
> I am not sure what the goals of conda-eda are

To provide continuously built version of EDA tools for the conda ecosystem.

> and if it makes sense to repackage VerilogAE if the wheels are already available on pypi.

It's not strictly necessarly (since conda environment.yml can also provision packages from PyPI), but repackaging with conda can ensure the coherence of other native dependencies with the rest of conda-eda packages.

> Building LLVM for OpenVAF can be tricky so it might sense to just use the official pre-build binary instead of compiling OpenVAF yourself.

If possible, I'd like to attempt the rebuild them using the conda toolchains, so that we can have nightly build and keep the ability to apply local patches to smooth the integration with the conda environment.

> The associated repo may help with setting up the build

Thanks for the reference, we'll likely use this as a reference while integrating with the existing of the conda-eda CI https://github.com/hdl/conda-eda/blob/master/.github/workflows/Build.yml.

>  I published an extremely old version of openvaf there years ago to reserve the name.

Thanks for the heads up!