hdl/conda-eda #281: klayout package should set KLAYOUT_PATH

proppy, 2023-01-25:
@erendn filed 

> I managed to find one of the issues. $KLAYOUT_PATH was pointing to another klayout installation, and klayout failed to load those plugins. Now, I'm getting the following error (klayout 0.28.3):
> 
> ```
> ERROR: In freepdk45.lydrc: 'source': Stream has unknown format: sram_bank.gds in Layout::read
> ERROR: 'source': Stream has unknown format: sram_bank.gds in Layout::read in Executable::execute
>   freepdk45.lydrc:13:in `execute'
>   :/built-in-macros/drc_interpreters.lym:27:in `instance_eval'
>   :/built-in-macros/drc_interpreters.lym:27:in `execute'
> ```

proppy, 2023-01-25:
@erendn filed 

> I fixed this issue by setting $KLAYOUT_PATH to `miniconda/lib/klayout`.
> 
> @proppy - I can reproduce the same issue with conda-eda/klayout if $KLAYOUT_PATH is unset.

erendn, 2023-01-25:
I realized that klayout searches for plugins in `miniconda/lib` but the plugins are copied to `miniconda/lib/klayout`. Is there a reason why you create this directory in `build.sh`?
```
mkdir -p ${PREFIX}/lib/klayout/
cp -ar pymod *_plugins ${PREFIX}/lib/klayout/
```