hdl/conda-eda #282: Klayout doesn't read gds properly

atorkmabrains, 2023-01-27:
Hi @proppy ,

Klayout built here doesn't seem to work properly.

I have checked the code in the following file:
https://github.com/hdl/conda-eda/blob/c8cc9a7e6975475da23e4a85b91cfc896ad2c480/misc/klayout/build.sh

To reproduce the issue:

```ruby
################################################################################################
# Copyright 2022 GlobalFoundries PDK Authors
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#     https://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.
################################################################################################

#===========================================================================================================================
#------------------------------------------- GF 0.18um MCU DRC RULE DECK --------------------------------------------------
#===========================================================================================================================
require 'time'
require "logger"

exec_start_time = Time.now

logger = Logger.new(STDOUT)

logger.formatter = proc do |severity, datetime, progname, msg|
  "#{datetime}: Memory Usage (" + `pmap #{Process.pid} | tail -1`[10,40].strip + ") : #{msg}
"
end

#================================================
#----------------- FILE SETUP -------------------
#================================================

# optional for a batch launch :   klayout -b -r gf_018mcu.drc -rd input=design.gds -rd report=gp180_drc.lyrdb

logger.info("Starting running GF180MCU Klayout DRC runset on %s" % [$input])
logger.info("Ruby Version for klayout: %s" % [RUBY_VERSION])

if $input
    if $topcell
        source($input, $topcell)
    else
        source($input)
    end
end

if $table_name
  table_name = $table_name
else
  table_name = "main"
end

logger.info("Loading database to memory is complete.")

if $report
    logger.info("GF180MCU Klayout DRC runset output at: %s" % [$report])
    report("DRC Run Report at", $report)
else
    logger.info("GF180MCU Klayout DRC runset output at default location." % [File.join(File.dirname(RBA::CellView::active.filename), "gf180_drc.lyrdb")])
    report("DRC Run Report at", File.join(File.dirname(RBA::CellView::active.filename), "gf180_drc.lyrdb"))
end

#================================================
#------------- LAYERS DEFINITIONS ---------------
#================================================
polygons_count = 0
logger.info("Read in polygons from layers.")

def get_polygons(l, d)
  if $run_mode == "deep"
    polygons(l, d)
  else
    polygons(l, d).merged
  end
end

comp           = get_polygons(22 , 0 )
count = comp.count()
logger.info("comp has %d polygons" % [count])
polygons_count += count

dnwell         = get_polygons(12 , 0 )
count   = dnwell.count()
logger.info("dnwell has %d polygons" % [count])
polygons_count  += count

nwell          = get_polygons(21 , 0 )
count   = nwell.count()
logger.info("nwell has %d polygons" % [count])
polygons_count  += count

lvpwell        = get_polygons(204, 0 )
count   = lvpwell.count()
logger.info("lvpwell has %d polygons" % [count])
polygons_count  += count

dualgate       = get_polygons(55 , 0 )
count   = dualgate.count()
logger.info("dualgate has %d polygons" % [count])
polygons_count  += count

poly2          = get_polygons(30 , 0 )
count   = poly2.count()
logger.info("poly2 has %d polygons" % [count])
polygons_count  += count

nplus          = get_polygons(32 , 0 )
count   = nplus.count()
logger.info("nplus has %d polygons" % [count])
polygons_count  += count

pplus          = get_polygons(31 , 0 )
count   = pplus.count()
logger.info("pplus has %d polygons" % [count])
polygons_count  += count

poly2.output("poly", "Poly")

exec_end_time = Time.now
run_time = exec_end_time - exec_start_time
logger.info("%s DRC Total Run time %f seconds" % [table_name, run_time])
```

Run command:
```bash
klayout -b -r example.drc -rd input=./test.gds -rd report=gp180_drc.lyrdb
```

atorkmabrains, 2023-01-27:
@proppy Another example of such failure in the regression CI:

https://github.com/efabless/globalfoundries-pdk-libs-gf180mcu_fd_pv/actions/runs/4024088674/jobs/6915711502

proppy, 2023-01-31:
@atorkmabrains thanks for reporting this issue, this seems similar to what was reported here:
https://github.com/hdl/conda-eda/issues/281#issuecomment-1404363267
https://github.com/KLayout/klayout/issues/1263#issuecomment-1405561058

PR #284 (2023-01-31): misc/klayout: fix plugins location
Fixes #282 #281
closed 2023-01-31 by proppy