hdl/conda-eda #283: Add git revisions as labels

donn, 2023-01-30:
Would it be possible to add labels that are like `gitrev-XXXXXXXXXXXXXXX` that have the full git revision of a specific build?

mithro, 2023-01-30:
FYI - @proppy

mithro, 2023-01-30:
@PiotrZierhoffer  is this information already available in some way?

proppy, 2023-01-30:
@donn git revision of the upstream repo, or the conda-eda repo?

If the former the short rev is already part of the archive name, prefixed with `g`, see:
https://anaconda.org/LiteX-Hub/openlane/files

donn, 2023-01-31:
That's not entirely an option, as there is no way I'm aware of to translate the long rev into `${version}_g${shortrev}` without cloning the repository myself first or maintaining a dictionary somewhere.

mithro, 2023-01-31:
@donn - What are you trying to do? The `{shortrev}` value should work anywhere the full revision value is accepted...

donn, 2023-01-31:
We don't have shortrev set up in the OpenLane tool Docker repo tags (`efabless/openlane-tools`), so we can't use the shortrevs there. Additionally, shortrevs are not immutable: in the (admittedly small) chance that another revision is generated with the first same 7 characters in the revision hash, the shortrev immediately becomes ambiguous: git will not generate the same shortrev for said revision nor will it accept it as a means of reaching said revision again.

Additionally, that does not solve the version and revision numbers as part of the version tag.

I was wondering if we could just have a direct mapping from the full revision hash to a label. It would make my life a lot easier.