hdl/conda-eda #286: add cosmopolitan build variant

proppy, 2023-02-04:
As discussed with @mithro and @hzeller, it could be interesting to have https://github.com/jart/cosmopolitan `noarch` build variant of some of the package so that they can easily be installed on multiple platform.

I think this could work for pure cli package like (yosys, magic, openroad, netgen, xls), graphical package or python package with native extension might be more challenging.

curious what others think?

mithro, 2023-02-04:
Great in theory, much harder to do in practice. I don't think we have the bandwidth to support yet another build flow?

mithro, 2023-02-04:
[YoWASP](https://yowasp.org/) might be a better / easier solution in the short term?