hdl/conda-eda #308: GDS factory integration

spnadig, 2023-03-08:
python writing on klayout does not work in this environment @joamatab
I followed steps in klayout integration and when I run klayout - I get:
[image]
running - gdsfactory 6.12.0
Looks like klayout used in this build is missing something?

proppy, 2023-03-09:
I suspect that the `--without-qt-binding` option here is to blame:
https://github.com/hdl/conda-eda/blob/0ff90c1af6b90f5b684ad558b134f8fde2859581/misc/klayout/build.sh#L35

proppy, 2023-03-09:
Also filed https://github.com/hdl/conda-eda/issues/309 to get gdsfactory packaged in conda-eda.