hdl/conda-eda #312: klayout grapics card issue in linux

spnadig, 2023-03-14:
"failed to load driver: swrast"

On running klayout - I get this warning
[image]

proppy, 2023-03-14:
@spnadig can you confirm which version of linux you're running with `uname -a`.

This seems related to:
https://github.com/ContinuumIO/anaconda-issues/issues/12889
https://github.com/conda-forge/ctng-compilers-feedstock/issues/95

spnadig, 2023-03-14:
Linux  5.19.11-1rodete1-amd64 #1 SMP PREEMPT_DYNAMIC Debian 5.19.11-1rodete1 (2022-10-31) x86_64 GNU/Linux

proppy, 2023-03-14:
This seems to be a version missmatch with the libc version used in the environment and the one the driver are linked against (with `LIBGL_DEBUG=verbose`):
```
libGL: MESA-LOADER: failed to open /usr/lib/x86_64-linux-gnu/dri/iris_dri.so: /home/proppy/silicon-gf180mcuC-env/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /lib/x86_64-linux-gnu/libLLVM-15.so.1)
```
And if forcing:
```
LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libstdc++.so.6 klayout
```
no errors.

proppy, 2023-03-14:
Updating https://anaconda.org/conda-forge/libstdcxx-ng locally seems to solve the issue by upgrading to `libstdc++.so.6` to a newer version (`12.2.0`).

The `main` channel https://anaconda.org/main/libstdcxx-ng is still on an older version (`11.2.0`).