hdl/conda-eda #313: SKY130 klayout -e not loading PDK layers.

njcoburn, 2023-03-23:
[image]
This error appears when opening klayout -e
When clicking the SKY130 PDK within klayout, also generates the following error. Layers of .gds are not named appropriately.
```
(base) nigelcoburn@godemperor:~/MS_SKY130$ klayout -e
ERROR: Reading /usr/local/google/home/nigelcoburn/MS_SKY130/silicon-env/share/pdk/sky130A/libs.tech/klayout/lvs/sky130.lylvs: XML parser error: not well-formed (invalid token) in line 31, column 48
ERROR: Reading /usr/local/google/home/nigelcoburn/MS_SKY130/silicon-env/share/pdk/sky130A/libs.tech/klayout/lvs/sky130.lylvs: XML parser error: not well-formed (invalid token) in line 31, column 48
libGL error: MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast
ERROR: /usr/local/google/home/nigelcoburn/.klayout/salt/gdsfactory/pymacros/klive.lym:19: module 'pya' has no attribute 'QTcpServer'
  /usr/local/google/home/nigelcoburn/.klayout/salt/gdsfactory/pymacros/klive.lym:19 (class AttributeError)
ERROR: /usr/local/google/home/nigelcoburn/MS_SKY130/silicon-env/share/pdk/sky130A/libs.tech/klayout/pymacros/cells/via_generator.py:23: No module named 'gdsfactory'
  /usr/local/google/home/nigelcoburn/MS_SKY130/silicon-env/share/pdk/sky130A/libs.tech/klayout/pymacros/cells/via_generator.py:23
  /usr/local/google/home/nigelcoburn/MS_SKY130/silicon-env/share/pdk/sky130A/libs.tech/klayout/pymacros/cells/vias.py:21
  /usr/local/google/home/nigelcoburn/MS_SKY130/silicon-env/share/pdk/sky130A/libs.tech/klayout/pymacros/cells/__init__.py:22
  /usr/local/google/home/nigelcoburn/MS_SKY130/silicon-env/share/pdk/sky130A/libs.tech/klayout/pymacros/sky130.lym:9 (class ModuleNotFoundError)
Warning: Initialization of layers failed: Unable to open file: /usr/local/google/home/nigelcoburn/.klayout/tech/sky130.lyp (errno=2)
Warning: Initialization of layers failed: Unable to open file: /usr/local/google/home/nigelcoburn/.klayout/tech/sky130.lyp (errno=2)
Warning: Initialization of layers failed: Unable to open file: /usr/local/google/home/nigelcoburn/.klayout/tech/sky130.lyp (errno=2)
```

proppy, 2023-03-31:
Seems similar to:
https://github.com/efabless/sky130_klayout_pdk/issues/3

proppy, 2023-03-31:
@njcoburn 

can you try to edit `$CONDA_PREFIX/share/pdk/libs.tech/klayout/tech/sky130A.lyt`, and replace those three elements:
```
 <base-path/>
 <original-base-path>$PDK_ROOT/$PDK/libs.tech/klayout</original-base-path>
 <layer-properties_file>sky130A.lyp</layer-properties_file>
```

njcoburn, 2023-03-31:
Copying the ```sky130A.lyp``` file from ```/usr/local/google/home/nigelcoburn/MS_SKY130/silicon-env/share/pdk/sky130A/libs.tech/klayout/tech``` to  ```/usr/local/google/home/nigelcoburn/.klayout/tech``` resolved the layer read issue as shown. 
[image]
I also edited the file ```sky130A.lyt``` file [directory: ```/usr/local/google/home/nigelcoburn/MS_SKY130/silicon-env/share/pdk/sky130A/libs.tech/klayout/tech```] line 9 to ```<layer-properties_file>sky130A.lyp</layer-properties_file>```

When running klayout -e I still get the following errors:
```(base) nigelcoburn@godemperor:~/MS_SKY130$ klayout -e
ERROR: Reading /usr/local/google/home/nigelcoburn/MS_SKY130/silicon-env/share/pdk/sky130A/libs.tech/klayout/lvs/sky130.lylvs: XML parser error: not well-formed (invalid token) in line 31, column 48
ERROR: Reading /usr/local/google/home/nigelcoburn/MS_SKY130/silicon-env/share/pdk/sky130A/libs.tech/klayout/lvs/sky130.lylvs: XML parser error: not well-formed (invalid token) in line 31, column 48
libGL error: MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast
ERROR: /usr/local/google/home/nigelcoburn/.klayout/salt/gdsfactory/pymacros/klive.lym:19: module 'pya' has no attribute 'QTcpServer'
  /usr/local/google/home/nigelcoburn/.klayout/salt/gdsfactory/pymacros/klive.lym:19 (class AttributeError)
ERROR: /usr/local/google/home/nigelcoburn/MS_SKY130/silicon-env/share/pdk/sky130A/libs.tech/klayout/pymacros/cells/via_generator.py:23: No module named 'gdsfactory'
  /usr/local/google/home/nigelcoburn/MS_SKY130/silicon-env/share/pdk/sky130A/libs.tech/klayout/pymacros/cells/via_generator.py:23
  /usr/local/google/home/nigelcoburn/MS_SKY130/silicon-env/share/pdk/sky130A/libs.tech/klayout/pymacros/cells/vias.py:21
  /usr/local/google/home/nigelcoburn/MS_SKY130/silicon-env/share/pdk/sky130A/libs.tech/klayout/pymacros/cells/__init__.py:22
  /usr/local/google/home/nigelcoburn/MS_SKY130/silicon-env/share/pdk/sky130A/libs.tech/klayout/pymacros/sky130.lym:9 (class ModuleNotFoundError)
```

proppy, 2023-07-12:
Related to https://github.com/efabless/sky130_klayout_pdk/issues/6