hdl/conda-eda #324: macosx package build failure for most packages

proppy, 2023-04-27:
It seems that most macosx package builds fails with the following error:
```
  SpecsConfigurationConflictError: Requested specs conflict with configured specs.
    requested specs: 
      - conda
    pinned specs: 
      - python=3.7
```
See:
https://github.com/hdl/conda-eda/actions/runs/4817053021

proppy, 2023-04-27:
Looks like 3.7 support has been dropped:
https://github.com/conda/conda/issues/12436

proppy, 2023-04-27:
The logs seems to stop after:
```
2023-04-27T16:53:01.0066342Z [32;1m16:53:01[0m | Prepare all required actions
2023-04-27T16:53:01.0255686Z ##[group]Run ./ci
2023-04-27T16:53:01.0256134Z [32;1m16:53:01[0m | env:
2023-04-27T16:53:01.0256487Z [32;1m16:53:01[0m |   ANACONDA_TOKEN: 
2023-04-27T16:53:01.0256846Z [32;1m16:53:01[0m |   ANACONDA_USER: 
2023-04-27T16:53:01.0257223Z [32;1m16:53:01[0m |   DEBIAN_FRONTEND: noninteractive
2023-04-27T16:53:01.0257615Z [32;1m16:53:01[0m |   GHA_MACHINE_TYPE: n2-standard-4
2023-04-27T16:53:01.0258073Z [32;1m16:53:01[0m |   APT_REQUIREMENTS: binutils build-essential csh curl git libgl1 pkg-config sudo wget
2023-04-27T16:53:01.0258517Z [32;1m16:53:01[0m |   PACKAGE: misc/magic
2023-04-27T16:53:01.0258866Z [32;1m16:53:01[0m |   OS_NAME: linux
2023-04-27T16:53:01.0259190Z ##[endgroup]
2023-04-27T16:53:01.1152545Z [32;1m16:53:01[0m | A local action modified in a branch!
2023-04-27T16:53:01.1387668Z [32;1m16:53:01[0m | Post job cleanup.
```
I wonder if this could be related to custom runners?

/cc @PiotrZierhoffer @ajelinski

spnadig, 2023-04-28:
This is currently effecting us for GF submission as I need to use magic version 8.3.396 in my conda env for extraction. Please let me know when a conda build with PDK updates and magic latest version is available from proppy/conda

proppy, 2023-05-01:
I wonder where th e error message `A local action modified in a branch!` comes from.

umarcor, 2023-05-02:
@proppy that message is coming from a branch that does not exist in this repo anymore. 

There seems to be a bug in Custom Runners, which picks the action YML from unrelated branches. #320 was created to debug this (which is based on #319). In #320, two jobs are executed on Default Runners, which behave as expected. Other jobs, the ones running on the Custom Runners, do fail.

Hence, a possible temporary workaround is to use Default Runners for the tools that need updating.

proppy, 2023-05-16:
@kgugala can you /cc the person looking into this?

proppy, 2023-05-17:
/cc @eszpotanski

eszpotanski, 2023-05-19:
We found the cause of this error and fixed it. Soon @AdamOlech will update Custom Runners, what should resolve this issue.

proppy, 2023-05-19:
@eszpotanski thanks!

can you link to the fix?

AdamOlech, 2023-05-19:
The [fix](https://github.com/antmicro/runner/commit/6de9ba8bee37de495912122674d586c8c56ff0c2) is on our end in the runner software so no changes here are required. I've just updated the coordinator running the jobs from this repository to the latest version so this bug should no longer occur.

umarcor, 2023-05-19:
@proppy I triggered a run of the master branch 20 min ago: https://github.com/hdl/conda-eda/actions/runs/5023776088.

It seems that the issue was fixed, as reported by @AdamOlech. However, during the last month, most of the jobs in the master branch were producing false positives. Since CI is now being properly executed, some legit Conda issue that was hidden is now breaking the builds:

```
2023-05-19T11:29:35.6486891Z [31;1m11:29:35[0m | UnsatisfiableError: The following specifications were found to be incompatible with a past
2023-05-19T11:29:35.6487973Z [31;1m11:29:35[0m | explicit spec that is not an explicit spec in this operation (setuptools):
```

umarcor, 2023-05-19:
Merging https://github.com/hdl/conda-eda/pull/326 fixed the error. Hence, I'm closing this as completed.

closed 2023-05-19 by umarcor