hdl/conda-eda #328: Document how to use conda-eda

xobs, 2023-05-02:
I was under the impression that conda-eda was usable from within conda, however it doesn't work, and there doesn't appear to be any documentation on how to use conda-eda under https://hdl.github.io/conda-eda/Usage.html

```
(base) user@Ondo:/si$ conda config --env --add channels conda-eda
(base) user@Ondo:/si$ conda upgrade -c conda-eda --all
Collecting package metadata (current_repodata.json): failed

UnavailableInvalidChannel: HTTP 404 NOT FOUND for channel conda-eda <https://conda.anaconda.org/conda-eda>

The channel is not accessible or is invalid.

You will need to adjust your conda configuration to proceed.
Use `conda config --show channels` to view your configuration's current state,
and use `conda config --show-sources` to view config file locations.

(base) user@Ondo:/si$
```

Please add documentation on how to use this repository as a user, rather than as someone building packages.

umarcor, 2023-05-02:
I think the channel is `litex-hub` due to historic reasons: https://github.com/search?q=repo%3Ahdl%2Fconda-eda%20litex-hub&type=code.

xobs, 2023-05-02:
That would explain some of the confusion I had.

I think it make sense to add that to the `Usage.html` guide, in addition to the README.md, so users can know how to use the packages built as part of this repo.

umarcor, 2023-05-23:
Refs:

- https://github.com/RTimothyEdwards/open_pdks/issues/376
- https://github.com/google/skywater-pdk/pull/421

/cc @proppy