hdl/conda-eda #329: open_pdks.sky130a package needs to be updated to latest version

michahn01, 2023-05-15:
As of this writing, the latest release of OpenPDKs is v1.0.414 (http://opencircuitdesign.com/open_pdks/history.html#Version%201_0). However, LiteX-Hub only supports v1.0.406 (https://anaconda.org/litex-hub/open_pdks.sky130a). 

When OpenPDKs.sky130a v1.0.406 is downloaded via Conda from LiteX-Hub and used with Magic, Netgen, OpenROAD, and Yosys (all downloaded from LiteX-Hub as well), OpenPDKs does not function correctly. However, when OpenPDKs v1.0.414 is installed manually (all else held constant), OpenPDKs works correctly. This issue is related to #idea-fasoc/OpenFASOC/issues/209.

I believe a fix might be to update the version of OpenPDKs on LiteX-Hub to the latest official release, v1.0.414.

mithro, 2023-05-15:
FYI - @proppy

proppy, 2023-05-15:
See #324

michahn01, 2023-05-29:
> See #324

#324 seems to have been resolved. @proppy When could we expect the version of open_pdks.sky130a to be updated?

proppy, 2023-06-01:
Looks like the timing test are failing:
```
**********************************************************************
File "$SRC_DIR/skywater-pdk/scripts/python-skywater-pdk/skywater_pdk/liberty.py", line 68, in __main__.TimingType
Failed example:
    (TimingType.leakage | TimingType.ccsnoise).names()
Expected:
    'basic, ccsnoise, leakage'
Got:
    'basic, leakage'
**********************************************************************
File "$SRC_DIR/skywater-pdk/scripts/python-skywater-pdk/skywater_pdk/liberty.py", line 71, in __main__.TimingType
Failed example:
    TimingType.ccsnoise.names()
Expected:
    'basic, ccsnoise'
Got:
    'basic'
**********************************************************************
1 items had failures:
   2 of  10 in __main__.TimingType
***Test Failed*** 2 failures.
```

proppy, 2023-06-02:
Seems like the failure is related to python `3.11`, as the tests passes with python `3.9` and `3.10`:
```
82 tests in 28 items.
82 passed and 0 failed.
Test passed.
```

proppy, 2023-06-02:
Looks like it's a behavior change in 3.11, see https://github.com/python/cpython/issues/99304

In particular the documentation now states that "only canonical flags are returned during iteration": https://docs.python.org/3/howto/enum.html#flag-and-intflag-minutia:~:text=only%20canonical%20flags%20are%20returned%20during%20iteration%3A

PR #336 (2023-06-02): misc/open_pdks/sky130a: use same python version as CI
Fixes #329 

Also filed https://github.com/google/skywater-pdk/issues/422 to fix it properly upstream.
closed 2023-06-02 by proppy