hdl/conda-eda #338: most of the linux build seems broken

proppy, 2023-06-26:
See https://github.com/hdl/conda-eda/actions/runs/5366668979

It doesn't show any failure logs so I suspect this has to do with custom runners, @kgugala @umarcor any clues?

kgugala, 2023-06-27:
I've seen some GitHub hiccup yesterday in a few other repos (I suspect this is an issue). I've restarted failed jobs and will check the logs

kgugala, 2023-06-27:
ok, seems to be back online - the runners are picking up jobs and running the builds

proppy, 2023-06-28:
Thanks! I'm now able to investigate the build failure for openroad and xls.

Created #339 and #340 to investigate.

proppy, 2023-07-17:
found the following error:
```
[2023-07-12 23:50:01Z ERR  GCPCoordinator] Traceback (most recent call last):
[2023-07-12 23:50:01Z ERR  GCPCoordinator]   File "vm_command.py", line 1140, in <module>
[2023-07-12 23:50:01Z ERR  GCPCoordinator]     main()
[2023-07-12 23:50:01Z ERR  GCPCoordinator]   File "/usr/lib/python3/dist-packages/click/core.py", line 764, in 
__call__
[2023-07-12 23:50:01Z ERR  GCPCoordinator]     return self.main(*args, **kwargs)
[2023-07-12 23:50:01Z ERR  GCPCoordinator]   File "/usr/lib/python3/dist-packages/click/core.py", line 717, in 
main
[2023-07-12 23:50:01Z ERR  GCPCoordinator]     rv = self.invoke(ctx)
[2023-07-12 23:50:01Z ERR  GCPCoordinator]   File "/usr/lib/python3/dist-packages/click/core.py", line 956, in 
invoke
[2023-07-12 23:50:01Z ERR  GCPCoordinator]     return ctx.invoke(self.callback, **ctx.params)
[2023-07-12 23:50:01Z ERR  GCPCoordinator]   File "/usr/lib/python3/dist-packages/click/core.py", line 555, in 
invoke
[2023-07-12 23:50:01Z ERR  GCPCoordinator]     return callback(*args, **kwargs)
[2023-07-12 23:50:01Z ERR  GCPCoordinator]   File "vm_command.py", line 1129, in main
[2023-07-12 23:50:01Z ERR  GCPCoordinator]     bucket_name, remove_uploaded
[2023-07-12 23:50:01Z ERR  GCPCoordinator]   File "vm_command.py", line 958, in upload_multiple_object_to_bucke
t
[2023-07-12 23:50:01Z ERR  GCPCoordinator]     next(destination_names) if destination_names else None)
[2023-07-12 23:50:01Z ERR  GCPCoordinator]   File "vm_command.py", line 925, in upload_object_to_bucket
[2023-07-12 23:50:01Z ERR  GCPCoordinator]     return r.status_code == 200, r.json()
[2023-07-12 23:50:01Z ERR  GCPCoordinator]   File "/usr/lib/python3/dist-packages/requests/models.py", line 897
, in json
[2023-07-12 23:50:01Z ERR  GCPCoordinator]     return complexjson.loads(self.text, **kwargs)
[2023-07-12 23:50:01Z ERR  GCPCoordinator]   File "/usr/lib/python3.7/json/__init__.py", line 348, in loads
[2023-07-12 23:50:01Z ERR  GCPCoordinator]     return _default_decoder.decode(s)
[2023-07-12 23:50:01Z ERR  GCPCoordinator]   File "/usr/lib/python3.7/json/decoder.py", line 337, in decode
[2023-07-12 23:50:01Z ERR  GCPCoordinator]     obj, end = self.raw_decode(s, idx=_w(s, 0).end())
[2023-07-12 23:50:01Z ERR  GCPCoordinator]   File "/usr/lib/python3.7/json/decoder.py", line 355, in raw_decode
[2023-07-12 23:50:01Z ERR  GCPCoordinator]     raise JSONDecodeError("Expecting value", s, err.value) from None
[2023-07-12 23:50:01Z ERR  GCPCoordinator] json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char
 0)
```
in `/home/runner/github-actions-runner/_layout/3__diag/Worker_20230712-234449-utc.log`, nto sure if that's related.