hdl/conda-eda #343: openlane package is broken

proppy, 2023-07-11:
```
2023-07-10T23:55:37.6853685Z [32;1m23:55:37[0m | [INFO]: Running Synthesis (log: ../_test_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placeh/share/openlane/designs/inverter/runs/RUN_2023.07.10_23.55.33/logs/synthesis/1-synthesis.log)...
2023-07-10T23:55:37.6854481Z [32;1m23:55:37[0m | [ERROR]: during executing yosys script $PREFIX/share/openlane/scripts/yosys/synth.tcl
2023-07-10T23:55:37.6855523Z [32;1m23:55:37[0m | [ERROR]: Log: ../_test_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placeh/share/openlane/designs/inverter/runs/RUN_2023.07.10_23.55.33/logs/synthesis/1-synthesis.log
2023-07-10T23:55:37.6856208Z [32;1m23:55:37[0m | [ERROR]: Last 10 lines:
2023-07-10T23:55:37.6856970Z [32;1m23:55:37[0m | ERROR: TCL interpreter returned an error: can't read "::env(SYNTH_CAP_LOAD)": no such variable
2023-07-10T23:55:37.6857471Z [32;1m23:55:37[0m | child process exited abnormally
2023-07-10T23:55:37.6857844Z [32;1m23:55:37[0m | 
2023-07-10T23:55:37.6858234Z [32;1m23:55:37[0m | [ERROR]: Creating issue reproducible...
```

proppy, 2023-07-11:
should resolve itself once https://github.com/The-OpenROAD-Project/OpenLane/pull/1892 is merged.

kareefardi, 2023-07-12:
@proppy does this means that conda builds of OpenLane doesn't use the version of dependencies listed here https://github.com/The-OpenROAD-Project/OpenLane/blob/master/dependencies/tool_metadata.yml ?

proppy, 2023-07-12:
@kareefardi yes, conda build everything from `HEAD` every might and only publish if the tests (hardening a test design) passes.

proppy, 2023-07-17:
build failed with the following error:
```
2023-07-17T15:26:18.1263165Z [32;1m15:26:18[0m | failed to get install actions, retrying.  exception was: Unsatisfiable dependencies for platform noarch: {'python_abi=3.7[build=*_cp37m]', "yaml[version='>=0.1.7,<0.2.0a0|>=0.2.5,<0.3.0a0']", "ncurses[version='>=6.1,<7.0a0|>=6.2,<7.0a0|>=6.3,<7.0a0|>=6.4,<7.0a0']", 'ca-certificates', 'feature:|@/linux-64::__glibc==2.31=0', "libgcc-ng[version='>=11.2.0|>=9.3.0']", "fontconfig[version='>=2.12.4,<3.0a0|>=2.12.6,<3.0a0|>=2.13.0,<3.0a0|>=2.14.1,<3.0a0']", "ncurses[version='6.0.*|>=6.0,<7.0a0|>=6.1,<7.0a0|>=6.2,<7.0a0|>=6.3,<7.0a0|>=6.4,<7.0a0']", "sqlite[version='>=3.20.1,<4.0a0|>=3.22.0,<4.0a0|>=3.23.1,<4.0a0|>=3.24.0,<4.0a0|>=3.25.2,<4.0a0|>=3.25.3,<4.0a0|>=3.26.0,<4.0a0|>=3.27.2,<4.0a0|>=3.29.0,<4.0a0|>=3.30.0,<4.0a0|>=3.30.1,<4.0a0|>=3.31.1,<4.0a0|>=3.32.3,<4.0a0|>=3.33.0,<4.0a0|>=3.35.4,<4.0a0|>=3.36.0,<4.0a0|>=3.37.2,<4.0a0|>=3.38.0,<4.0a0|>=3.39.3,<4.0a0|>=3.40.0,<4.0a0|>=3.40.1,<4.0a0|>=3.41.1,<4.0a0|>=3.41.2,<4.0a0|>=3.39.2,<4.0a0|>=3.38.3,<4.0a0|>=3.38.2,<4.0a0']", "qt[version='>=5.9.7,<5.10.0a0']", "jpeg[version='>=9b,<10a']", "__glibc[version='>=2.17']", "qt[version='>=5.0,<6.0|>=5.9.7,<5.10.0a0']", "sqlite[version='>=3.24.0,<4.0a0|>=3.25.2,<4.0a0|>=3.25.3,<4.0a0|>=3.26.0,<4.0a0|>=3.27.2,<4.0a0|>=3.29.0,<4.0a0|>=3.30.1,<4.0a0|>=3.31.1,<4.0a0|>=3.33.0,<4.0a0|>=3.35.4,<4.0a0|>=3.36.0,<4.0a0|>=3.38.0,<4.0a0|>=3.39.3,<4.0a0|>=3.40.0,<4.0a0|>=3.40.1,<4.0a0|>=3.41.1,<4.0a0|>=3.41.2,<4.0a0']", "libgcc-ng[version='>=7.2.0|>=7.3.0|>=7.5.0']", "libxml2[version='>=2.10.3,<2.11.0a0|>=2.9.14,<2.10.0a0|>=2.9.10,<2.10.0a0']", 'feature:/linux-64::__glibc==2.31=0', "ncurses[version='>=6.2,<7.0a0|>=6.3,<7.0a0']", "_openmp_mutex[version='>=4.5|>=5.1']", 'xorg-libx11', "sqlite[version='>=3.20.1,<4.0a0|>=3.22.0,<4.0a0|>=3.23.1,<4.0a0|>=3.24.0,<4.0a0|>=3.25.2,<4.0a0|>=3.25.3,<4.0a0|>=3.26.0,<4.0a0|>=3.27.2,<4.0a0|>=3.29.0,<4.0a0|>=3.30.1,<4.0a0|>=3.31.1,<4.0a0|>=3.33.0,<4.0a0|>=3.35.4,<4.0a0|>=3.36.0,<4.0a0|>=3.38.0,<4.0a0|>=3.39.3,<4.0a0|>=3.40.0,<4.0a0|>=3.40.1,<4.0a0|>=3.41.1,<4.0a0|>=3.41.2,<4.0a0|>=3.37.2,<4.0a0|>=3.32.3,<4.0a0|>=3.30.0,<4.0a0|>=3.39.2,<4.0a0|>=3.38.3,<4.0a0|>=3.38.2,<4.0a0']", "fontconfig[version='>=2.12.4,<3.0a0|>=2.12.6,<3.0a0|>=2.13.0,<3.0a0']", "fontconfig[version='>=2.13.1,<3.0a0|>=2.13.96,<3.0a0']", "sqlite[version='>=3.20.1,<4.0a0|>=3.22.0,<4.0a0|>=3.23.1,<4.0a0|>=3.24.0,<4.0a0|>=3.25.2,<4.0a0|>=3.25.3,<4.0a0|>=3.26.0,<4.0a0|>=3.27.2,<4.0a0|>=3.29.0,<4.0a0|>=3.30.1,<4.0a0|>=3.31.1,<4.0a0|>=3.33.0,<4.0a0|>=3.35.4,<4.0a0|>=3.36.0,<4.0a0|>=3.38.0,<4.0a0|>=3.39.3,<4.0a0|>=3.40.0,<4.0a0|>=3.40.1,<4.0a0']", "sqlite[version='>=3.20.1,<4.0a0|>=3.22.0,<4.0a0|>=3.23.1,<4.0a0|>=3.24.0,<4.0a0|>=3.25.2,<4.0a0|>=3.25.3,<4.0a0|>=3.26.0,<4.0a0|>=3.27.2,<4.0a0|>=3.29.0,<4.0a0|>=3.30.1,<4.0a0|>=3.31.1,<4.0a0|>=3.33.0,<4.0a0|>=3.35.4,<4.0a0|>=3.36.0,<4.0a0|>=3.37.2,<4.0a0|>=3.38.0,<4.0a0|>=3.39.3,<4.0a0|>=3.40.0,<4.0a0|>=3.40.1,<4.0a0|>=3.41.1,<4.0a0|>=3.41.2,<4.0a0|>=3.39.2,<4.0a0|>=3.38.3,<4.0a0|>=3.38.2,<4.0a0|>=3.32.3,<4.0a0|>=3.30.0,<4.0a0']", "libgcc-ng[version='>=11.2.0']", "sqlite[version='>=3.24.0,<4.0a0|>=3.25.2,<4.0a0|>=3.25.3,<4.0a0|>=3.36.0,<4.0a0|>=3.37.2,<4.0a0|>=3.38.0,<4.0a0|>=3.39.3,<4.0a0|>=3.40.0,<4.0a0|>=3.40.1,<4.0a0|>=3.41.1,<4.0a0|>=3.41.2,<4.0a0|>=3.35.4,<4.0a0|>=3.33.0,<4.0a0|>=3.31.1,<4.0a0|>=3.30.1,<4.0a0|>=3.29.0,<4.0a0|>=3.27.2,<4.0a0|>=3.26.0,<4.0a0']", "libgcc-ng[version='>=11.2.0|>=7.5.0|>=7.3.0|>=7.2.0']", "libgcc-ng[version='>=11.2.0|>=9.3.0|>=7.5.0|>=7.3.0|>=7.2.0']", "libgcc-ng[version='>=7.2.0|>=7.5.0']", "fontconfig[version='>=2.13.0,<3.0a0']", "bzip2[version='>=1.0.8,<2.0a0']", "sqlite[version='>=3.24.0,<4.0a0|>=3.25.2,<4.0a0|>=3.25.3,<4.0a0|>=3.26.0,<4.0a0|>=3.27.2,<4.0a0|>=3.29.0,<4.0a0|>=3.30.1,<4.0a0|>=3.31.1,<4.0a0|>=3.33.0,<4.0a0|>=3.35.4,<4.0a0|>=3.36.0,<4.0a0|>=3.38.0,<4.0a0|>=3.39.3,<4.0a0|>=3.40.0,<4.0a0|>=3.40.1,<4.0a0|>=3.41.2,<4.0a0|>=3.32.3,<4.0a0|>=3.30.0,<4.0a0']", "sqlite[version='>=3.24.0,<4.0a0|>=3.25.2,<4.0a0|>=3.25.3,<4.0a0|>=3.26.0,<4.0a0|>=3.27.2,<4.0a0|>=3.29.0,<4.0a0|>=3.30.0,<4.0a0|>=3.30.1,<4.0a0|>=3.31.1,<4.0a0|>=3.32.3,<4.0a0|>=3.33.0,<4.0a0|>=3.35.4,<4.0a0|>=3.36.0,<4.0a0|>=3.37.2,<4.0a0|>=3.38.0,<4.0a0|>=3.39.3,<4.0a0|>=3.40.0,<4.0a0|>=3.40.1,<4.0a0|>=3.41.1,<4.0a0|>=3.41.2,<4.0a0|>=3.39.2,<4.0a0|>=3.38.3,<4.0a0|>=3.38.2,<4.0a0']", "libxml2[version='>=2.9.8,<2.10.0a0']", "_libgcc_mutex[version='*|0.1',build=main]", "yaml[version='>=0.2.5,<0.3.0a0']", "libgcc-ng[version='>=11.2.0|>=9.3.0|>=7.3.0']", "_openmp_mutex[version='>=4.5']", "libxml2[version='>=2.9.7,<2.10.0a0|>=2.9.8,<2.10.0a0']", "ncurses[version='>=6.1,<7.0a0|>=6.4,<7.0a0']", '_openmp_mutex'}
2023-07-17T15:26:18.1272940Z [32;1m15:26:18[0m | WARNING:conda_build.build:failed to get install actions, retrying.  exception was: Unsatisfiable dependencies for platform noarch: {'python_abi=3.7[build=*_cp37m]', "yaml[version='>=0.1.7,<0.2.0a0|>=0.2.5,<0.3.0a0']", "ncurses[version='>=6.1,<7.0a0|>=6.2,<7.0a0|>=6.3,<7.0a0|>=6.4,<7.0a0']", 'ca-certificates', 'feature:|@/linux-64::__glibc==2.31=0', "libgcc-ng[version='>=11.2.0|>=9.3.0']", "fontconfig[version='>=2.12.4,<3.0a0|>=2.12.6,<3.0a0|>=2.13.0,<3.0a0|>=2.14.1,<3.0a0']", "ncurses[version='6.0.*|>=6.0,<7.0a0|>=6.1,<7.0a0|>=6.2,<7.0a0|>=6.3,<7.0a0|>=6.4,<7.0a0']", "sqlite[version='>=3.20.1,<4.0a0|>=3.22.0,<4.0a0|>=3.23.1,<4.0a0|>=3.24.0,<4.0a0|>=3.25.2,<4.0a0|>=3.25.3,<4.0a0|>=3.26.0,<4.0a0|>=3.27.2,<4.0a0|>=3.29.0,<4.0a0|>=3.30.0,<4.0a0| [... cut ...] 4.0a0|>=3.37.2,<4.0a0|>=3.38.0,<4.0a0|>=3.39.3,<4.0a0|>=3.40.0,<4.0a0|>=3.40.1,<4.0a0|>=3.41.1,<4.0a0|>=3.41.2,<4.0a0|>=3.39.2,<4.0a0|>=3.38.3,<4.0a0|>=3.38.2,<4.0a0']", "libxml2[version='>=2.9.8,<2.10.0a0']", "_libgcc_mutex[version='*|0.1',build=main]", "yaml[version='>=0.2.5,<0.3.0a0']", "libgcc-ng[version='>=11.2.0|>=9.3.0|>=7.3.0']", "_openmp_mutex[version='>=4.5']", "libxml2[version='>=2.9.7,<2.10.0a0|>=2.9.8,<2.10.0a0']", "ncurses[version='>=6.1,<7.0a0|>=6.4,<7.0a0']", '_openmp_mutex'}
```

proppy, 2023-07-21:
for some reason the build is picking up a super old version of `open_pdks.sky130a` now:
```
2023-07-20T23:59:19.6007907Z [32;1m23:59:19[0m |     open_pdks.sky130a: 1.0.286_0_g52af776-20220221_020327         ***                           
```

proppy, 2023-07-21:
I suspect this is due to `open_pdks.sky130A` tests taking a long time and not being ready by the time openlane tests kick in.

proppy, 2023-07-21:
Still a failure on python 3.10 which I believe is due to https://github.com/The-OpenROAD-Project/OpenROAD/issues/2535

PR #348 (2023-07-23): misc/openlane: improve pdk testing
- github/workflows: add open_pdks jobs dependencies (required by tests)
- misc/openlane: add gf180mcuC test
- misc/openlane: set python version from ci
- remove ubuntu 22.04 requirements (as python comes from the conda install)

Hopefully fixes #343
closed 2023-07-23 by proppy