hdl/conda-eda #354: Who is maintaining "Conda EDA tools packages for Aarch64?

jun1okamura, 2023-09-04:
Hi

CC: @proppy 

I am trying to install OpenEDA/PDK into my MAC M2 on Ubuntu(aarch64) 22.04 by UTM but seems there are some missing packages. 

## Steps to Reproduce the Problem

> make timing
 
> ...
 
> Retrieving notices: ...working... done
> Collecting package metadata (repodata.json): done
> Solving environment: failed
> 
> ResolvePackageNotFound:
> 
> netlistsvg
> iverilog
> yosys
> make: *** [/home/okamura/PDK/skywater-pdk/third_party/make-env/conda.mk:145: /home/okamura/PDK/skywater-pdk/env/conda/envs/skywater-pdk-scripts/bin/python] Error 1
> 

## Specifications
### Version:
### Platform: 
Ubuntu (Aarch64) 22.04 on MAC M2 by UTM (Version 4.2.5)

proppy, 2023-09-04:
Can you confirm if you're trying to install on MacOSX on Ubuntu?

I think @xobs packaged a few dependencies for MacOSX natively.

jun1okamura, 2023-09-04:
Thanks, and do you mean below?

https://github.com/xobs/conda-eda

proppy, 2023-09-04:
@jun1okamura I mean that @xobs contributed the MacOSX arm package to https://github.com/hdl/conda-eda/, see https://github.com/hdl/conda-eda/pulls?q=is%3Apr+author%3Axobs+

so we should be able to discuss with him what was covered by his recent work.

jun1okamura, 2023-09-04:
Hi @proppy,

I think MacOSX native is fine for me, but Ubuntu on UTM can completely separate its environment. Welcome to discuss it and I would like to contribute somehow if possible.

xobs, 2023-09-06:
I did manage to get it to build locally, and had a native end-to-end flow going. The only trick was that there was a bug in openroad where the environment was getting reinitialised, but I believe that's solved in the latest version.

The problem I ran into was that I couldn't figure out how to cross-compile it, and there weren't any arm64 builders available for free. One potential solution is to move to conda-forge which appears to have the machinery in place to cross-compile, but that hasn't been done yet.

jun1okamura, 2023-09-11:
@xobs
I'm sorry for not getting back to you sooner and thank you for sharing it. Can I confirm that the most reasonable way to use the arm64 platform is by building a native environment on my Mac or Ubuntu using the openroad source?

xobs, 2023-09-11:
I believe the "official" way is to run the Ubuntu images via Docker. But as I've had a lot of trouble with Docker in the past, my preference is to run things natively.

jun1okamura, 2023-09-14:
@xobs 
Finally, I figured out how your repo works, built "conda-eda" Aarch64 Linux packages for my Ubuntu on Mac M2 by UTM, and uploaded them into my Anaconda account. Thank you for your advice. 
Jun

jun1okamura, 2023-09-25:
@xobs
I had trouble building some of the modules, such as Klayout, and magic... as the following. It seems to me related to 'Jinja2' and 'cdt'. Appreciate it if you could let me know any advice.
Jun
---
[image]

jun1okamura, 2023-10-04:
Finally, I got a proper way to build Klayout on Aarch64 for my M2 MAC. Thanks!

[image]