hdl/conda-eda #355: miniconda3 releases doesn't support py3.7 but openroad, magic and netgen conda packages are pinned to py3.7

saicharan0112, 2023-09-11:
I am trying to update openroad to the latest package with miniconda3 updated to 23.5.* but it looks like openroad, magic and netgen are pinned to python 3.7, and latest miniconda3 doesn't have a release for py3.7 .

Can I please know if this is the packages issue or the conda installation issue?

[image]

proppy, 2023-09-11:
@saicharan0112 it should be available for other version of python as well, see: https://anaconda.org/LiteX-Hub/openroad/files

saicharan0112, 2023-10-01:
@proppy sorry I finally got some time to work on this. I think I found the reason behind this issue. Its that the python version that comes with conda is not aligned ? https://github.com/automl/auto-sklearn/issues/1240 This man had the same issue and installing the right version again worked for him. Now I tried that same and able to install all packages for 3.10. Thanks for your help @proppy :)

closed 2023-10-01 by saicharan0112