hdl/conda-eda #357: openroad binary library issue

saicharan0112, 2023-10-03:
I installed openroad conda package (openroad-2.0_10145_g0de6f5131) but looks like the binary is missing some libraries. 
I am running it on ubuntu:22 with conda 23.9.0. 

/usr/bin/miniconda3/bin/openroad: error while loading shared libraries: libfmt.so.8: cannot open shared object file: No such file or directory
Command exited with non-zero status 127

saicharan0112, 2023-10-03:
Yesterday I managed to install it. Not sure what went wrong while installing the conda packages. I will debug more and write it down here just in case if someone might come across this

chetanyagoyal, 2023-10-07:
> Yesterday I managed to install it. Not sure what went wrong while installing the conda packages. I will debug more and write it down here just in case if someone might come across this

Fixed the reproduced error by running - 
`sudo apt install libfmt-dev`

closed 2023-10-23 by saicharan0112